mpostol/OPC-UA-OOI #212: Problem with build entire solution

AdamBorowiecki, 2017-07-16:
After git clone and build entire solution there is 5 errors:

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\UANodeSetValidation\bin\Debug\UAOOI.SemanticDataUANodeSetValidation.dll' could not be found	SemanticData.AddressSpaceComplianceTestTool	F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\AddressSpaceComplianceTestTool\CSC	1	Active

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS1513	} expected	SemanticData.UANodeSetValidation	F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\UANodeSetValidation\BuildErrors1.cs	8	Active

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\AddressSpaceComplianceTestTool\bin\Debug\AddressSpaceComplianceTestTool.exe' could not be found	SemanticData.AddressSpaceComplianceTestTool.UnitTests	F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\Tests\AddressSpaceComplianceTestToolUnitTests\CSC	1	Active

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\UANodeSetValidation\bin\Debug\UAOOI.SemanticDataUANodeSetValidation.dll' could not be found	SemanticData.DataSerialization.UnitTest	F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\Tests\DataSerializationUnitTestProject\CSC	1	Active

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0006	Metadata file 'F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\UANodeSetValidation\bin\Debug\UAOOI.SemanticDataUANodeSetValidation.dll' could not be found	SemanticData.UANodeSetValidation.UnitTest	F:\Workspace\Clone_MP_GH\OPC-UA-OOI\SemanticDataSolution\Tests\USNodeSetValidationUnitTestProject\CSC	1	Active

mpostol, 2017-07-17:
It must be investigated on the computer hosting the clone, because it may be caused by the different location of the .sln file. In original solution the projects refers to the folder `packages` located in the same folder as the.sln. Adding this project to different solution doesn't cause reference path change, so it still refers to the same folder, but it is empty.  Open the `UAObjectOrientedInternet.sln`, populate the nugget cache for this solution. This should solve the problem.

AdamBorowiecki, 2017-07-18:
I do not add this project to another solution - it was just git clone <repo>.

 Maybe it could be connected with below error with open project UANetworkingArchitecture
.modelproj? With Visual Studio are you using? 2015 or 2017?

Unsupported
This version of Visual Studio is unable to open the following projects. The project types may not be installed or this version of Visual Studio may not support them. 
For more information on enabling these project types or otherwise migrating your assets, please see the details in the "Migration Report" displayed after clicking OK.
	 - UANetworkingArchitecture, "F:\Workspace\clone_MP\OPC-UA-OOI\CommonResources\ModelingProject\UANetworkingArchitecture\UANetworkingArchitecture.modelproj"

No changes required
These projects can be opened in Visual Studio 2015, Visual Studio 2013, Visual Studio 2012, and Visual Studio 2010 SP1 without changing them.
	 - SemanticData.UANodeSetValidation, "F:\Workspace\clone_MP\OPC-UA-OOI\SemanticDataSolution\UANodeSetValidation\SemanticData.UANodeSetValidation.csproj"
	 - .CommonResources, ".CommonResources"
	 - SemanticData.InformationModelFactory, "F:\Workspace\clone_MP\OPC-UA-OOI\SemanticDataSolution\InformationModelFactory\SemanticData.InformationModelFactory.csproj"
....

mpostol, 2017-07-18:
2015 - porting to 2017 is on the road map, but if project is not added to your solution it should not generate compilation error, except you are building cloned solution using original .sln. 
What it exactly means `After git clone and build entire solution....` - which one ?

mpostol, 2017-08-11:
Some projects depend on the .NET 3.5 that is missing on your computer. I will upgrade the solution up to 4.61  for all projects.  I hope your project doesn't depend on these projects, the errors are not relevant.

mpostol, 2017-08-13:
the problem with: `UANetworkingArchitecture, "F:\Workspace\clone_MP\OPC-UA-OOI\CommonResources\ModelingProject\UANetworkingArchitecture\UANetworkingArchitecture.modelpro`
is cased because `This version of Visual Studio is unable to open the following projects` - I am using Enterprise.
This problem cannot have side effect on others - you can simply neglect it.

mpostol, 2017-10-14:
I have ported the solution to VS 2017 see #214 . The code is error free. Hope it helps.

closed 2017-10-14 by mpostol