mpostol/OPC-UA-OOI #213: Build errors after add OPC-UA-OOI as submodule

AdamBorowiecki, 2017-07-16:
There is 2 error after add OPC-UA-OOI as submodule to another projects:

**Error nr 1**

> Severity	Code	Description	Project	File	Line	Suppression State
> Error		This project references NuGet package(s) that are missing on this computer. Enable NuGet Package Restore to download them.  For more information, see http://go.microsoft.com/fwlink/?LinkID=322105. The missing file is F:\Workspace\Integration-MQTT-with-ObjectOrintedInternet\\.nuget\NuGet.targets.	
> 
> Project: Configuration.DataBindings	
> 
> F:\Workspace\Integration-MQTT-with-ObjectOrintedInternet\OPC-UA-OOI\Configuration\DataBindings\Configuration.DataBindings.csproj	144
> 

**Error 2**

> Severity	Code	Description	Project	File	Line	Suppression State
> Error		This project references NuGet package(s) that are missing on this computer. Enable NuGet Package Restore to download them.  For more information, see http://go.microsoft.com/fwlink/?LinkID=322105. The missing file is F:\Workspace\Integration-MQTT-with-ObjectOrintedInternet\\.nuget\NuGet.targets.	
> 
> Project: Configuration.DataBindings.UnitTest	
> 
> F:\Workspace\Integration-MQTT-with-ObjectOrintedInternet\OPC-UA-OOI\Configuration\Tests\DataBindingsUnitTest\Configuration.DataBindings.UnitTest.csproj	137

mpostol, 2017-08-11:
I have tested it and it seems that the answer is in the message: `Enable NuGet Package Restore to download them`.

mpostol, 2017-10-14:
Leave a comment

closed 2017-10-14 by mpostol