mpostol/OPC-UA-OOI #224: Networking.ReferenceApplication - problem with build

AdamBorowiecki, 2017-11-18:
There is build error:
The type or namespace 'CommonServiceLocator' could be not found.

mpostol, 2017-11-18:
The Nuget package must be updated. Current copy in repo uses 2.0.1. Previous version had different namespace.

AdamBorowiecki, 2017-11-19:
Yes, I handle it. I only want to suggest that we should be able to build project after git pull(without any additional action).

mpostol, 2017-11-19:
The repo is updated and refers only to the new version, so after cloning the new version of nuget packages are downloaded. I have tested it.

I am not sure what happens after pulling for existing copy because the current version referenced is in the packages folder that is ignored for pushing. 

Anyway, now the repo refers only to the new version, and the code had been fixed accordingly. 

I assume that we can close this issue without any additional action.

closed 2017-11-19 by mpostol