mpostol/OPC-UA-OOI #227: Problem to include this projects to another solution

AdamBorowiecki, 2017-11-22:
Integration of this project with other solution(for example: https://github.com/AdamBorowiecki/Integration-MQTT-with-ObjectOrintedInternet) is problematic, because after integration we get a lot of build error. The cause is lack of "package" directory in OPC-UA-OOI. 

To archive this directory we need to compile UAObjectOrientedInternet.sln first, not for example: Integration-MQTT-with-OOI.sln.

mpostol, 2017-11-24:
Yes, you are right, because each project looks for packages using relative path against calculated using original solution file location. Adding the same project to another solution doesn't change this path, therefore packages cannot be correctly localized.
I have no idea if there is something simple we can do about it. The provided workaround has drawback that we need direct references to .dll files instead of references to projects, but fortunately it works.

Maybe someone knows how to help.

mpostol, 2017-11-24:
I hope #226 should solve this problem finally. 

Maybe the solution is to use [packages.config](https://docs.microsoft.com/en-us/nuget/schema/packages-config)

Let me know how it works for you.

mpostol, 2018-03-21:
After adding nuget packager there is no need to merge the source code

closed 2018-03-21 by mpostol