mpostol/OPC-UA-OOI #245: Port the library UAOOI.Networking.SemanticData to .NET STandard

mpostol, 2018-03-13:
# Description

## Scope

- use [.NET Portability Analyzer](https://marketplace.visualstudio.com/items?itemName=ConnieYau.NETPortabilityAnalyzer) to check the library against portability
- modify the code
- improve testing

## Related work

- planned Xamarin HMI implementation

mpostol, 2018-03-22:
To properly manage NuGet packages and avoid UT errors related to the wrong version new solutions must be defined.  It must be converged with this work.

mpostol, 2018-03-23:
The first priority is to port the projects:
- `Networking.SemanticData`
- `Configuration.Networking`
- `Common.Infrastructure`

Question for users - what requirements must be met to port these libraries? 
E.g.  Does .NET Standard 2.0  meet all requirements to create Xamarin HMI mobile implementation?

Drutol, 2018-03-24:
Hello, I've successfully built several Xamarin apps using .NET Standard libraries targeting profile:
```xml
  <PropertyGroup>
    <TargetFramework>netstandard2.0</TargetFramework>
  </PropertyGroup>
```
Porting to .Net Standard is the most prospective course of action right now, it'll be possible to run the library within .Net Core 2.0 apps, Xamarin apps etc.

mpostol, 2018-03-24:
Move it to separate project as it is not related to WPF.

commit c17f06ec (2018-06-19): Port the library UAOOI.Networking.SemanticData to .NET STandard fixes #245
- The library has been ported.
-UT green

OPCUAServerProducerSimulator - move to separate project fixes #262
- new project SimulatorInteroperabilityTest created

API changes:

IValueConverter.cs - added instead of referring to WPF stuff
closed 2018-06-18 by mpostol