mpostol/OPC-UA-OOI #253: I can not start UAOOI.Networking.ReferenceApplication.3.0.1-Echo

AdamBorowiecki, 2018-04-25:
After double click on UAOOI.Networking.ReferenceApplication the circle is spinning and there is no action.

mpostol, 2018-04-25:
Try  [3.0.5](https://github.com/mpostol/OPC-UA-OOI/releases) added to 3.0.0 prerelease.. It provides more information in the log. If any problem occurs report once more and add the log file to the issue by dragging and dropping.

AdamBorowiecki, 2018-04-29:
3.0.5 also not working. I can not even start app, so there is no log. I try to start as user and as administration. In both cases I get 2 attached pictures.
[image]
[image]

Drutol, 2018-04-29:
Hello, same thing happens to me.
Some logs:
```
Application: UAOOI.Networking.ReferenceApplication.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NotSupportedException

Exception Info: System.IO.FileLoadException
   at System.Reflection.RuntimeAssembly._nLoad(System.Reflection.AssemblyName, System.String, System.Security.Policy.Evidence, System.Reflection.RuntimeAssembly, System.Threading.StackCrawlMark ByRef, IntPtr, Boolean, Boolean, Boolean)
   at System.Reflection.RuntimeAssembly.nLoad(System.Reflection.AssemblyName, System.String, System.Security.Policy.Evidence, System.Reflection.RuntimeAssembly, System.Threading.StackCrawlMark ByRef, IntPtr, Boolean, Boolean, Boolean)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(System.Reflection.AssemblyName, System.Security.Policy.Evidence, System.Reflection.RuntimeAssembly, System.Threading.StackCrawlMark ByRef, IntPtr, Boolean, Boolean, Boolean)
   at System.Reflection.Assembly.Load(System.Reflection.AssemblyName)
   at System.ComponentModel.Composition.Hosting.AssemblyCatalog.LoadAssembly(System.String)
   at System.ComponentModel.Composition.Hosting.AssemblyCatalog..ctor(System.String)
   at UAOOI.Networking.ReferenceApplication.AppBootstrapper.ConfigureContainer()
   at UAOOI.Networking.ReferenceApplication.MEF.MefBootstrapper.Run(Boolean)
   at UAOOI.Networking.ReferenceApplication.AppBootstrapper.Run(Boolean)
   at UAOOI.Networking.ReferenceApplication.App.App_Startup(System.Object, System.Windows.StartupEventArgs)
   at System.Windows.Application.OnStartup(System.Windows.StartupEventArgs)
   at System.Windows.Application.<.ctor>b__1_0(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at System.Windows.Threading.DispatcherOperation.InvokeInSecurityContext(System.Object)
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at MS.Internal.CulturePreservingExecutionContext.Run(MS.Internal.CulturePreservingExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndWrapper.WndProc(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(System.Windows.Threading.DispatcherPriority, System.TimeSpan, System.Delegate, System.Object, Int32)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr, Int32, IntPtr, IntPtr)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(System.Windows.Interop.MSG ByRef)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Threading.Dispatcher.PushFrame(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Application.RunDispatcher(System.Object)
   at System.Windows.Application.RunInternal(System.Windows.Window)
   at System.Windows.Application.Run(System.Windows.Window)
   at UAOOI.Networking.ReferenceApplication.App.Main()
```

```
Faulting application name: UAOOI.Networking.ReferenceApplication.exe, version: 3.0.5.0, time stamp: 0x5abfea69
Faulting module name: KERNELBASE.dll, version: 10.0.16299.371, time stamp: 0xee297154
Exception code: 0xe0434352
Fault offset: 0x00103ef2
Faulting process ID: 0x1b14
Faulting application start time: 0x01d3dfb9f4a53adc
Faulting application path: I:\OPC\New Echo\UAOOI.Networking.ReferenceApplication.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report ID: f4450955-c868-4a9f-95d5-0947bb1d25c1
Faulting package full name: 
Faulting package-relative application ID: 
```

mpostol, 2018-04-29:
As it refers to current release I am reopening this issue.

mpostol, 2018-04-29:
I have confirmed the problem on the Azure WServer Virtual machine. The problem is a result of locking *.exe and *.dll files after unzipping bu operating system. 

## Workaround:

For all mentioned above files open properties one by one and unlock the file. It helps on WS 2012 and W7.

Solutions;

1. use scripts to unlock the file - not sure if it is applicable
2. Prepare.msi  - this kind of projects is not available in VS2017 - we must look for an extension
3. Use different packing software: RAR , 7ZIP,  etc.
4. Use Click Once -  we need hosting server to expose the installation stuff and code sign certificate is available. My certificate expires on 12th May.
5. Keep as is. 

Let me know what do you think.  
Let keep it open until fixing the problem.

AdamBorowiecki, 2018-04-29:
Ok, I use 7-ZIP and I can lauch program without problem. I use Windows 10 Pro.

mpostol, 2018-06-08:
I have added self-extracting file:
[UAOOI.Networking.ReferenceApplication.3.0.5.exe](https://github.com/mpostol/OPC-UA-OOI/releases/download/3.0.0/UAOOI.Networking.ReferenceApplication.3.0.5.exe)

It is available for [3.0.0 Release.](https://github.com/mpostol/OPC-UA-OOI/releases/tag/3.0.0)

Hope it solves the problem.

closed 2018-06-08 by mpostol