mpostol/OPC-UA-OOI #318: Access to imported NodeSet

FrederickBrier, 2018-12-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

FrederickBrier, 2018-12-21:
The AddressSpaceContext class is sealed and while it provides an ImportUANodeSet method, it populates a set of private dictionaries that are not directly accessible, or possible to iterate over. The goal is to create a new C# client code generator that will utilize the UANodeSet.xsd compliant XML files exported by the UA Modeler.

The following are some of the issues that are suggesting the need for a new client generator. The UA Model Compiler takes a UA Model Design.xsd compliant XML file. Further, it appears that the lower level Session.Read(), not Session.ReadValue(), must be used to avoid BadAttributeIdInvalid exceptions caused by un-populated values. We are considering writing a code generator to create classes, for the OPC model, whose interfaces do not contain OPC class types, and hide the OPC communications. This does not appear to be a feature of the UA Model Compiler, as it uses a generic PropertyState<> around primitives for a model classes's properties. One would expect that when these classes are added to the message context factory as an encodeable type, that they would be instantiated when using ReadValue(). Read() results are placed in a DataValueCollection. It isn't clear how to easily marshal the values back into the classes generated by the model compiler.

The thin layer of UANodeContext and UAModelContext around the UANodeSet XML serialization classes are internal and could not be traversed by code generation templates. Is there a different mechanism for accessing the node set model? Or is there a design consideration that required the model to be opaque? Thank  you.

mpostol, 2019-01-12:
@FrederickBrier thank you for your engagement and proposals.  In general, I agree that this topic is very important and the need to address the problems:
- Address Space creation
- Address Space behavior

I understand that the main purpose for the classes you are going to generate is a custom implementation of the AS behavior, e.g. process data update. 

Let's address your comments step by step.

> `The AddressSpaceContext class is sealed and while it provides an ImportUANodeSet method, it populates a set of private dictionaries that are not directly accessible, or possible to iterate over. [...]

The AddressSpaceContext class is really sealed and while it provides an ImportUANodeSet method, it populates a set of private dictionaries, but `IModelFactory` defines a generalized implementation of the specific representation of the OPC UA Information Model. To export the model during validation the interface must be implemented and the implementation injected in:  `AddressSpaceContext.InformationModelFactory`.
I am using this approach in the Address Space Model Designer to import and recover/recreate the ModelDesign xml files used by the tool and the ModelCompiler for UANodeSet.

I have added the #319 issue to address this problem. 

> `The following are some of the issues that are suggesting the need for a new client generator. The UA Model Compiler takes a UA Model Design.xsd compliant XML file. Further, it appears that the [...]

I am not sure about what do you have in mind referring to the Session.Read(), not Session.ReadValue(). I will appreciate more information and explanation in this regard.

> ` We are considering writing a code generator to create classes, for the OPC model, whose interfaces do not contain OPC class types, and hide the OPC communications. This does not appear to be a [...]

I am pretty sure that the UAOOI.SemanticData.UANodeSetValidation is a good starting point to follow up.  

> `The thin layer of UANodeContext and UAModelContext around the UANodeSet XML serialization classes are internal and could not be traversed by code generation templates. Is there a different [...]

See the explanation above. I am also considering to implement a new API for the `AddressSpaceContext` to provide full in-memory functionality of the Address Space according to the specification.  We can use it in any application Client/Server/PubSub to represent the process state and behaviour, i.e.to create in-memory underlaing process partial replica. 

Many thanks for feedback. Let me know how it works for you.

mpostol, 2019-01-12:
On more hint. While implementing

mpostol, 2019-03-14:
It will be addressed in:
- #319 
- #335
Now it must be recognized as duplicated.

closed 2019-03-14 by mpostol