mpostol/OPC-UA-OOI #384: Review the documentation: Getting Started Tutorial

mpostol, 2019-10-28:
[Getting Started Tutorial](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/READMEGettingStartedTutorial.md).

mpostol, 2019-11-17:
part of #394

210342, 2019-11-17:
I think it is worth to mention that `IEncodingFactory` implementation can be found in NuGet package `UAOOI.Networking.Encoding`. I have used it in my application. Also it would be helpful to provide an example of schema for a producer/consumer configuration file.

mpostol, 2019-11-17:
@210342 thanks for comments. I understand that the proposals can be recognized as future requests and are not a blocker for your work.

mpostol, 2019-11-21:
Add information about dependency on `commonservicelocator`.  It is used to ger parts required to compose the application.

mpostol, 2019-11-21:
There must be added that each instance of the `DataManagementSetup` creates a new instance of selected Roles. The `DataManagementSetup` can create consumer, producer of both depending on the configuration.

mpostol, 2019-12-04:
## `commonservicelocator` research on
`commonservicelocator` see also:

- [Microsoft.Practices.ServiceLocation](https://github.com/Drutol/CrossHMI)
- [Martin Fowler; Inversion of Control Containers and the Dependency Injection pattern](https://martinfowler.com/articles/injection.html)
- [Service Locator is an Anti-Pattern by Mark Seemann](https://blog.ploeh.dk/2010/02/03/ServiceLocatorisanAnti-Pattern/)

Drutol, 2019-12-04:
Hmm...
> in not .NET API Browser

Neither `Microsoft.Practices.ServiceLocation` is there.

> depends on Microsoft.Extensions.DependencyInjection

Not sure why would you want to use the whole package.
https://www.nuget.org/packages/Microsoft.Extensions.DependencyInjection.Abstractions/

> implements container ServiceCollection

Abstractions do not implement this class.

> don't agree - there are a lot of errors and misunderstandings

Well, the main point made there is that it obscures dependencies required by the components. The API of the library is not self descriptive and requires digging through the documentation to find out what's up. In my opinion that's very valid point, class is requesting something at random instead of getting the thing from its constructor.

> don't require a concrete container
> can be easily implemented for any container

These apply to the `IServiceCollection` too.

Just wanted to comment on that, I'm not really hung up on removing ServiceLocator but wanted to clarify a few things.

commit dd40a4d8 (2019-12-04): Review the documentation: Getting Started Tutorial #384

- fixes #384
- improved documentation
commit d29a63fc (2019-12-04): Review the documentation: Getting Started Tutorial #384

- fixes #384
- fixed typos
PR #405 (2019-12-04): Networking merge to master
Fixes #384
closed 2019-12-04 by mpostol