mpostol/OPC-UA-OOI #401: Custom configuration - review the documentation

mpostol, 2019-11-26:
It is related to Drutol/CrossHMI#7

Prepare document how to:

- create custom configuration
- implement dynamic configuration
- use configuration ion the custom application

Drutol, 2019-11-30:
I actually have one concern with providing custom configuration in a dynamic fashion. Assuming we have multiple devices with various configs the end library user should have an ability to distinguish which configuration was used for this particular binding/repository.
Considering:
```cs
    IConsumerBinding GetConsumerBinding(
      string repositoryGroup,
      string processValueName,
      UATypeInfo fieldTypeInfo);
```
It only provides two `strings` for data association. Without parsing the configuration on his own the user won't be able to match these against anything given that there are multiple configs.

This method could either provide reference to applicable configuration instance (but that means dependency on `ConfiguartionData`) or some token/identifier which library could generate upon adding configuration type. Additionally with token scenario it could come bundled with some kind of lifetime handle which when disposed would remove configuration type from the library. These are my first ideas that came to mind.

mpostol, 2019-11-30:
@Drutol, Many thanks for valuable comments. 

By design, the configuration is static now. Dynamic configuration is just a feature request on my road map. In the static approach, both mentioned above parameters come directly from the configuration file so they are under the application author (i.e. library user) control. At the very beginning, the `repositoryGroup` may be used to define the semantic-context, but the `processValueName` represents variable in this context.  `GetConsumerBinding` is called just after starting the application -  it is a step of application bootstrap procedure.  

My plan is to work on the documentation and provide How to... examples in the UT. After that, we will decide about the next steps. 

My concern is if dynamic configuration addressing the scenario when data in the NetworkMessage cannot be associated with the predefined semantics in the configuration file is really useful. Lets thing on a scenario where this feature is required. 

I am not sure what you mean `Assuming we have multiple devices with various configs..` If the device is boiler we shall not care about its configuration, but we need its type to define the semantics of the incoming values in the message. In the message, we have only fields that can be used to describe the source of data, i.e. `PublisherId` and `WriterGroupId`.  For details visit the [ABNF message syntax definition](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/MessageHandling/NetworkMessage.abnf).
In my opinion, more research is required addressing the dynamic data discovery process.

Drutol, 2019-11-30:
> Dynamic configuration is just a feature request on my road map.

Yes, I just thought it'd be a good place to add my comment since the issue contains
`* implement dynamic configuration`

> I am not sure what you mean `Assuming we have multiple devices with various configs.`

What I meant here is that if we have "discovered" 3 kinds devices we are ending up with 3 configurations where each contains encoding infrmation + for example image data with value mappings on it. I'm getting ahead of myself here with assuming configuration not being static.

> Lets think on a scenario where this feature is required.

Well, if we don't assume otherwise we are bound to the device set present at the application deployment. If we know them all beforehand it defeats the purpose of data discovery.

> In my opinion, more research is required addressing the dynamic data discovery process.

Yes of course. I've been just playing around with getting my code to handle possible `GetConsumerBinding` calls after bootstrapping process is finished and application is in the midst of its lifetime.

mpostol, 2019-12-02:
@Drutol 

Thanks for comments. 

2. Referring to

> What I meant here is that if we have "discovered" 3 kinds devices we are ending up with 3 configurations where each contains encoding infrmation + for example image data with value mappings on it. I'm getting ahead of myself here with assuming configuration not being static.

I must say it is not true. We don't know where the data comes from, so it is impossible to get the configuration - it is PubSub. The configuration must be derived by the RxNetworkin application from the Type of data using message content only. It could happen that three devices will have the same type. 

2. I have improved the documentation on how to use this library to implement the custom configuration. I will appreciate reviewing it and reporting any problems as separate issues referring to your code. 

3. Let's close this issue and merge it to master.  In case further discussion is required, post it on [Gitter](https://gitter.im/mpostol/OPC-UA-OOI)

commit ee83cd33 (2019-12-02): Custom configuration - review the documentation #401

- Improved documentation
- fixes #401
closed 2019-12-02 by mpostol