mpostol/OPC-UA-OOI #407: DataSetConfiguration.is not provided by the IBindingFactory

mpostol, 2019-12-09:
The problem is reported by 210210342/OPC-UA-OOI-webapp#42
Review how this interface is implemented in Drutol/CrossHMI#16

The workout is proposed in OPC-UA-OOI-webap - new entries in the configuration defining aliases to be used instead of `InformationModelURI `. The `InformationModelURI` is required to select data semantics against the types.

Drutol, 2019-12-09:
For the record "210210342/OPC-UA-OOI-webapp#42" is not quite available as it's private repo most probably.

mpostol, 2019-12-09:
The `210210342/OPC-UA-OOI-webapp#42` is really in the private repository, but the problem addressing configuration against type definitions is similar.

mpostol, 2019-12-09:
Take note that it is assigned to me. I must investigate the problem to learn how you have overtaken it.