mpostol/OPC-UA-OOI #465: Generate MachineVision ModelDesign format from NodeSet2 format

eoursel, 2020-08-16:
Hi @mpostol  

When i generate the ModelDesign format for [Machine Vision specification](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/MachineVision/Opc.Ua.MachineVision.NodeSet2.xml) i notice that many namespaces are created in the model design specification. The output file contains up to 180 namespaces such as at the end of the file

`<ObjectType xmlns:q180="http://opcfoundation.org/UA/MachineVision" SymbolicName="q180:VisionSystemType">`

Not sure this behavior is expected? Is there any reason to do that? 
[Opc.Ua.MachineVision.ModelDesign.xml.zip](https://github.com/mpostol/OPC-UA-OOI/files/5081156/Opc.Ua.MachineVision.ModelDesign.xml.zip)

mpostol, 2020-08-16:
Hi @eoursel, Thanks for your feedback. I will follow up on in the mpostol/ASMD#158. There is a master plan to test this stuff against existing models. It is a good opportunity to report more models to be tested. Check out the rules [ASMD Rel 5.0 OPC UA Information Models Testing Partnership Program](https://mpostol.github.io/ASMD/ASMD50-ModelsTesting.html)
Now I am working on Azure integration, but the modeling validation and conversion will be my next target.  Follow my projects to be informed about the progress.

mpostol, 2021-02-03:
Hi @eoursel,

I returned to work on Information Models. You will be available shortly to discuss your issue.

Is it still valid?.
Mariusz

eoursel, 2021-02-03:
hi @mpostol  
Mariusz, my concern about NodeSet2 to ModelDesigner conversion is always valid. I am really busy this week doing a OPC UA training, but i think i will have some timeslots next week to talk about that.  
Regards

Eric

mpostol, 2021-02-14:
## Testing scope

- [ ] possibility to open ModelDesign file by ASMD
- [ ] possibility to import NodeSet file
- [ ] validation against the schema and compliance with the specification (Part 3 and Part 5) of the NodeSet file
- [ ] possibility to convert NodeSet to ModelDesign
- [ ] possibility to compile ModelDesign and generate NodeSet file (round trip test)
- [ ] correct generation of the model documentation in MS Word format
- [ ] correct generation of website content with the model documentation

mpostol, 2021-02-14:
@eoursel  current results of import and validation are as follows. Let me know if there is something unexpected. The testing will be continued according to the scope presented above.

```XML
Test Name:	MachineVisionTest
Test Outcome:	Passed
Result StandardOutput:	
Debug Trace:
Trace: Verbose, Error Focus:Diagnostic, ErrorID: P0-0003010000 Info: It is diagnostic information Entering IAddressSpaceContext.ValidateAndExportModel - starting for the http://opcfoundation.org/UA/MachineVision namespace.
Trace: Verbose, Error Focus:Diagnostic, ErrorID: P0-0003010000 Info: It is diagnostic information AddressSpaceContext.ValidateAndExportModel - selected 60 nodes to be added to the model.
Trace: Verbose, Error Focus:Diagnostic, ErrorID: P0-0003010000 Info: It is diagnostic information Finishing Validator.ValidateExportModel - the model contains 60 nodes.
````