mpostol/OPC-UA-OOI #467: Implement communication to Azure

mpostol, 2020-08-17:
## Scope

- [ ] AzureCommunicationContext
- [ ] AzureCommunicationStateBase
- [ ] AzureCommunicationRegitering
- [ ] AzureCommunicationCommunication
- [ ] AzureCommunicationDisconnected

>Note: Out-of-band communication must be implemented using Data Transfer Object concept

mpostol, 2020-08-31:
Coniseder note [IAsyncDisposable](https://github.com/mpostol/OPC-UA-OOI/commit/34123d82ed0535b072347d5803c942531abc40f3#r41917701)

Drutol, 2020-09-01:
One thing to note about `IAsyncDisposable` is that it's netstandard2.**1** feature. So it is advised to implement both patterns.
Not sure about blocking the thread in synchronous variant though.

mpostol, 2020-09-04:
> One thing to note about `IAsyncDisposable` is that it's netstandard2.**1** feature. So it is advised to implement both patterns.
> Not sure about blocking the thread in synchronous variant though (may be worth closing the connection is separate method altogether for sync variant).

IDisposable is not needed anymore.  The state machine is responsible to dispose of all resources.

closed 2020-09-05 by mpostol