mpostol/OPC-UA-OOI #513: ObjectsFolder does not show other components than the server.

fgolra, 2021-02-07:
I have defined a folder object in my information model and imported the nodeset2 of this model in the ASMD v4.3.  If I organize this folder under the objectsfolder, the tool does not show this folder under "objects" in the browse view panel.

mpostol, 2021-02-07:
Transferred because the relevant code is provided as NuGet from this repository.

mpostol, 2021-02-07:
Further development will be conducted on the branch <https://github.com/mpostol/OPC-UA-OOI/tree/SemanticData-6.1.3>

mpostol, 2021-02-14:
## Scope for further testing:

- [ ] possibility to open ModelDesign file by ASMD
- [ ] possibility to import NodeSet file
- [ ] validation against the schema and compliance with the specification (Part 3 and Part 5) of the NodeSet file
- [ ] possibility to convert NodeSet to ModelDesign
- [ ] possibility to compile ModelDesign and generate NodeSet file (round trip test)
- [ ] correct generation of the model documentation in MS Word format
- [ ] correct generation of website content with the model documentation

mpostol, 2021-02-14:
@fgolra current testing results for the ValidateAndExportModel are as follows. Let me know if something is not as expected.

```TXT

Test Name:	fgolra177Test
Test Outcome:	Passed
Result StandardOutput:	
Debug Trace:
Trace: Verbose, Error Focus:Diagnostic, ErrorID: P0-0003010000 Info: It is diagnostic information Entering IAddressSpaceContext.ValidateAndExportModel - starting for the https://agileo-automation.com/UA/Semi/ namespace.
Trace: Verbose, Error Focus:Diagnostic, ErrorID: P0-0003010000 Info: It is diagnostic information AddressSpaceContext.ValidateAndExportModel - selected 1 nodes to be added to the model.
Trace: Verbose, Error Focus:Diagnostic, ErrorID: P0-0003010000 Info: It is diagnostic information Finishing Validator.ValidateExportModel - the model contains 1 nodes.
```