mpostol/OPC-UA-OOI #653: The exported model doesn't contain all nodes

mpostol, 2022-03-19:
## Describe the bug

Not all nodes are recognized following references starting from the selected to export nodes. Some nodes are not members of the complex types. Additionally, some nodes may be duplicated if they are targeted by many references.

## To Reproduce

Analyze the result of `UAObjectTypeTestMethod` and recover the DI model

## Tasks

- [ ] selects all the nodes belonging to the namespace chosen and create a dictionary `AllNodesInConcern`
- [ ] during exporting using an existing algorithm remove exported nodes from the `AllNodesInConcern`
- [ ] export the nodes that have been not removed
- [ ] Add one method AddorReturnNodeFactory with parameter key of int type
- [ ] Add key generator in case `AddNodeFactory` is used or there is a key conflict
- [ ] Move abstract implementation from validator to the modedel factory

## Additional context

It contributes to:

- #629
- mpostol/UA-Nodeset#51

mpostol, 2022-03-19:
@eoursel let me inform you that I have created a new more general issue to fix reported by your one. It is not only a problem with DI. The problem must be investigated using dedicated Unit Tests but not the whole model. I will get back to the reported issue after fixing this.

eoursel, 2022-04-09:
Hi @mpostol Sorry for the late answer. i am aware that the DI is just an example. I took the DI because this is one of the model where OPC Foundation publishes the NodeSet format and the ModelDesign format. Most of the new companion specifications do not contain the ModelDesign format. I am currently interested by the IEC61850 companion specification more than the DI. If you have already done the translation to ModelDesign for IEC61850 let me know. Nice weekend.

mpostol, 2022-04-09:
Hi @eoursel. We are still working on recovering ModelDesign for all available models using the current version of the asp. Check out the folders ModelDesign added to all models in the repository [mpostol/UA-Nodeset](https://github.com/mpostol/UA-Nodeset). 

Models have been generated thanks to @mertyusaatag contribution. He is working on an independent fork:

[mertyusaatag/UA-Nodeset](https://github.com/mertyusaatag/UA-Nodeset)

The final results are proposed by him as pr to my repository. 

I have just checked that the model you requested is not created for some reason😒. I have notified @mertyusaatag by adding a comment to the appropriate issue.: mertyusaatag/UA-Nodeset#11. It was already closed so the problem needs investigation. He promised to investigate the problem in detail next week. 

BTW let me stress that the current version of the asp (`ASP 6.5.0-Kilo`) still has the problem with adding all nodes to the recovered model. The problem is more complex than I expected previously. One of the problems is that models are not stable and even if the modification is unimportant it is hard to handle it during recovering ModelDesign while removing the inheritance chain. I still believe that it is possible but need time and motivation to fix it. Anyway, the workaround is to add just a few nodes if any manually to the ModelDesigne in the meantime.
I will try to mitigate the problem by independent development of the [mpostol/UA-ModelCompiler](https://github.com/mpostol/UA-ModelCompiler) with the goal to add it to ASP and ASMD tools as a nugget package.

mpostol, 2022-04-11:
## ModellingRule

|     id | Name                   |
| -----: | ---------------------- |
|     78 | `Mandatory`            |
|     80 | `Optional`             |
|     83 | `ExposesItsArray`      |
| 11 505 | `OptionalPlaceholder`  |
| 11 510 | `MandatoryPlaceholder` |